FluenTech/embedded-time #7: Signed/Unsigned Durations

PTaylor-FluenTech, 2020-05-16:
I began this journey after finding that the core::Duration was an unsigned value. I then investigated the `time` crate and found that while it does have signed Duration, it is focused more toward "wall-clock" time considerations. This led me to great this project.

I began my development with the use-case of a non-wrapping timer (typically chaining two 32-bit peripheral timers in a microcontroller). However, if this project is to be compatible with wrapping timers, that may drastically change its design.

Signed durations can be useful when calculating the difference between two instances. It allows the subtraction to be done in either order returning a logical result. For example, in the expression `x - y`, if `x` is _earlier_ than `y`, a negative duration is returned. However, if wrapping timers are to be supported, a negative duration will never be returned as subtracting a "higher-value" instance from a  "lower-value"  assumes the timer wrapped, and a positive duration is returned.

PTaylor-FluenTech, 2020-05-18:
> However, if wrapping timers are to be supported, a negative duration will never be returned as subtracting a "higher-value" instance from a "lower-value" assumes the timer wrapped, and a positive duration is returned.

This is inaccurate. In fact, a signed result of the wrapping arithmetic is important. Otherwise, Instances cannot be ordered.

PTaylor-us, 2020-06-21:
If it is absolutely necessary for the difference between two `Instant`s to result in a signed `Duration` (which I believe it is), could other types still utilize unsigned inner types? At the lowest level, the hardware timers are virtually always treated as unsigned values. What gets messy is when support for that timer overflowing from the max value back around to 0. By using a _signed_ wrapping_sub of the two `Instant`'s inner values, we are able to account for that overflow (as long as the the actual difference between them is less than `i32::MAX` ticks).

Perhaps the inner types could be unsigned and only be cast to signed in the case of getting the difference of `Instant`s. That would make `Clock` and `Instant` inner types unsigned, but `Duration`s would still be signed.

closed 2020-07-03 by PTaylor-us

devjgm, 2023-08-01:
Why did you change from signed to unsigned in https://github.com/FluenTech/embedded-time/pull/22? It sounded like in this issue you were saying that signed durations are good. I would strongly agree that Durations should be signed. Algebraically, durations _can_ be negative, so it's surprising that (by default) I can't create a duration that represents "1 minute ago". I.e., I can subtract a Duration from an Instant, but I can't add a negative Duration: `I - D` is OK, but `I + -D` is not?

PR #22 (2020-07-01): Change all time-related struct inner types to unsigned
Fixes #19 
Closes #7
PR #28 (2020-07-03): V0.6 Release
PRs to be included: #27, #21, #22 
Closes #23
Closes #7 
Closes #19