FluenTech/embedded-time #18: Add lossless duration handling within a given `Clock` domain

PTaylor-FluenTech, 2020-06-20:
# Motivation
Given the following call (which will not compile) where `instant` and `now` are two instants from `SysClock`:
```rust
SysClock::delay(instant.duration_since(&now).unwrap());
```
`SysClock` _may_ have any `Period`/precision, but the call of `duraction_since()` requires a `Duration` implementing type be specified. This is not optimal, particularly in generic or library situations as it requires an arbitrary decision about the precision.

PTaylor-FluenTech, 2020-06-20:
The immediate idea is to make the default duration one with the same `Period` as the Clock associated with the two instants. However, this would mean a less efficient duration scheme. Currently all `Period`s are associated `const`s from the `Duration` trait.

PTaylor-FluenTech, 2020-06-20:
My current line of investigation is adding `Instant` and `Clock` methods that eliminate the explicit conversion into a `Duration`. The methods would work with the raw ticks while maintaining the same `Period` (that of the `Clock`). However, these semantics may be too unwieldy and unintuitive to use.

PTaylor-us, 2020-07-12:
With the new `Timer` type taking over blocking delays, it now looks like this:
```rust
clock.new_timer(1_u32.seconds()).start()?.wait()?;
```
Where the goal would be to be able to leave the duration units unspecified.

Perhaps an API of `clock.delay_until(Instant)` would be doable. However, that may just push the problem lower down.

closed 2020-07-24 by PTaylor-us