FluenTech/embedded-time #26: Non-blocking interface for `Clock::delay()`?

eldruin, 2020-06-29:
I think it could be useful for the `Clock::delay()` method to have a non-blocking interface like: `fn delay<Dur>(dur: Dur) -> nb::Result<(), ()>`.
This would return `Err(nb::Error::WouldBlock)` as long as the delay is ongoing and `Ok(())` once finished.

This would give the user more control on polling and what to do while waiting and is the usual interface for other blocking methods (see [here](https://docs.rs/embedded-hal/0.2.4/embedded_hal/timer/trait.CountDown.html#required-methods) for example).

However, I am unsure if it would be possible to implement this in the provided `Clock::delay()` method implementation since the start would need to be kept constant until completion.

PTaylor-us, 2020-06-29:
Thank you for the issue. I am currently working on the next minor-version release which should include a `Timer` type. Feel free to take a look at the PR #21 and let me know what you think. It does allow for starting a timer, then either blocking until it expires (`wait()`) or polling. It also allows reading both elapsed and remaining time (in real-world duration units, of course). In fact, that PR will remove `Clock::delay()` to use the `Timer` functionality instead.

PTaylor-us, 2020-07-04:
I'm going to look into this as I have virtually no experience with `nb`. I'm hesitant to adopt semantics like this due to the uncertainty and deadlock I perceive in the embedded WG. If anything, I may add this alongside the current `wait()` interface.