FluenTech/embedded-time #40: Feature parity with bitrate crate?

little-arhat, 2020-07-12:
Hello!

I currently time/frequency/bitrate types from [bitrate crate](https://github.com/dfrankland/bitrate). I saw few discussions around this crate in embedded-hal and rtic-rs repos and decided to give it a try. However, some features from bitrate crate are absent from embedded-time:
* bitrate types (Bps);
* extensions types for numerics (`32.hz`);
* `Copy` is not implemented.

Do you want to make this crate go-to library for all frequency/time/bitrate related needs or should I stick with bitrate?

Thanks.

PTaylor-us, 2020-07-12:
Thank you so much for the feedback and interested in `embedded-time`. First, it is my intention to try to cover all time/frequency/bitrate related uses in embedded systems. 

Now your bullet points:
- bitrate types (Bps)
  - When I added the `frequency` module and `Hertz` type, I considered adding other types such as bps. At that time, I decided to hold off since units like bps, baud, etc are all fundamentally frequencies (or maybe more accurately, rate-based). It was my expectation that any specific rate-based units could be very simply aliased or wrap the Hertz type. If that is sub-optimal, I am definitely open to adding a separate Bps type. At that point I would probably also add a Baud type. Maybe some others. Open to suggestions.
- extensions types for numerics (32.hz)
  - Actually, this does exist for the `Hertz` type in the trait `NumericConstructor` which is implemented for `u32` and `u64` types. This trait can be included anonymously by `use`ing the `traits` module (`traits::*`). It looks like `123_u32.hertz()`.
- Copy is not implemented
  - That's my bad. I thought that `Copy` was automatically `impl`ed if all data members of the struct were `Copy`. I will get that fixed.

One question I have for you is how much value would there be to add _scaled_ frequency types (kilohertz, megahertz, etc). It would certainly add a bit of complexity to add those types in the same way the duration-based types are implemented. At the time, I felt that hertz would be sufficient (especially for embedded systems).

little-arhat, 2020-07-13:
@PTaylor-us thanks for the reply.

> It was my expectation that any specific rate-based units could be very simply aliased or wrap the Hertz type
stm32-rs [defines Bps](https://github.com/search?q=org%3Astm32-rs+struct+bps&type=Code) multiple times, and there is nothing specific in their definition. So even if they were defined as aliases to `Hertz` crates would still had to contain duplicate definitions, which is unfortunate.

Thanks for pointing me to `NumericConstructor`, will check it out. Personally, I think `hz` is sufficient.

PTaylor-us, 2020-07-13:
@little-arhat I will go ahead and add a concrete Bps type including extension construction.

closed 2020-07-24 by PTaylor-us