FluenTech/embedded-time #43: Separation of features into separate crates

PTaylor-us, 2020-07-14:
It may be necessary to split this crate as there are potentially different use-cases for various features.

PTaylor-us, 2020-07-14:
My initial impression is that the following discrete crates might make sense:

- time
  - `Duration` trait and off-the-shelf duration types
  - Rate types
- clock
  - `Clock` trait
  - `Instant` type
  - Dependencies
    - time
- timer
  - `Timer` type
  - Dependencies
    - clock

Largely, these dependencies are already realized. However the `Clock` trait contains a method `new_timer()` to spawn --- you guessed it, a new `Timer`. I like the composition given this structure:

```rust
let timer = clock.new_timer(2.seconds()).start();
```

If the timer functionality is separated from the `Clock` trait, we're left with:
```rust
let timer = Timer::new(my_clock, 2.seconds()).start();
```
Definitely not nearly as pretty. Any suggestions?

hannobraun, 2020-11-13:
> If the timer functionality is separated from the `Clock` trait, we're left with:
> 
> ```rust
> let timer = Timer::new(my_clock, 2.seconds()).start();
> ```
> 
> Definitely not nearly as pretty. Any suggestions?

Add an extension trait to `timer` (pseudo-code, not compiled or tested):
``` rust
pub trait ClockExt { // alternative name: `NewTimer`
    fn new_timer(duration) -> Timer { ... }
}

impl<T> ClockExt for T where T: Clock {}
```

That would require the user to import `ClockExt` to use that method, but that can be eased by creating a `prelude` module and telling users to always `use timer::prelude::*`.