FluenTech/embedded-time #47: Error trait is not easy to use in external device driver implementations

eldruin, 2020-07-19:
I implemented `Clock::now()` on my [ds323x](https://github.com/eldruin/ds323x-rs) I2C/SPI RTC driver.
The implementation was quite straight forward, the whole thing is in this file: [embedded_time.rs](https://github.com/eldruin/ds323x-rs/blob/embedded-time-no-error-trait/src/embedded_time.rs)
And here is an example: [embedded_time_linux.rs](https://github.com/eldruin/ds323x-rs/blob/embedded-time-no-error-trait/examples/embedded_time_linux.rs)

However, this was only this easy after the removal of `embedded_time::Error`. Drivers are generic over the underlying peripheral error types. In this case I2C and I/O pin errors. Normally no requirements are made on these generic types and the error type looks like this for example:
```rust
#[derive(Debug)]
pub enum Error<CommE, PinE> {
    /// I²C/SPI bus error
    Comm(CommE),
    /// Pin setting error
    Pin(PinE),
    /// Invalid input data provided
    InvalidInputData,
}
```

Since the `Error` trait requires `Debug` and `Eq`, this requirement needs to be [propagated](https://github.com/eldruin/ds323x-rs/blob/838f71447ca40b2206a9568ff34947b8c2d09ce2/src/embedded_time.rs#L32) to the generic error types and this is not very good because these are implementation-defined and are sometimes not implemented, making it suddenly impossible to use the driver if the MCU error implementation does not implement `Eq`, [for](https://github.com/stm32-rs/stm32f4xx-hal/blob/b5f21572cd0eb8b9654acb9cc9e5664f3da1c467/src/i2c.rs#L550) [example](https://github.com/rust-embedded/rust-i2cdev/blob/5bf63c4f6c447f179af951dc7a5ccaa43534f150/src/linux.rs#L37).

PTaylor-us, 2020-07-20:
That makes a lot of sense. I'll take a look at your PR and likely merge it. Thanks!

eldruin, 2020-07-20:
Great, thank you!

closed 2020-07-20 by eldruin