FluenTech/embedded-time #72: Duration calculated from clock instants cannot be used with concrete Durations

ryan-summers, 2020-08-10:
When a calculation is performed using two `Instant`s from a given `Clock`, the resulting `Duration` is of type `duration::Generic<Clock::T>`. It appears that this type does not implement the necessary `Try`, `TryFrom`, `Into`, or `TryInto` traits to be used for a comparison against a concrete duration, such as `duration::Seconds(1)`.

Example:
```rust

fn test<C: Clock>(clock: C) {
    let start = clock.try_now().unwrap();
    let end = clock.try_now().unwrap();

    let duration = start.checked_duration_since(&end).unwrap();

    let duration_secs: Result<Seconds<C::T>, _> = duration.try_into();
    if duration_secs.unwrap() > Seconds(1) {
        println!("Greater than 1 second);
    }
}

```

The result of attempting to convert the `Generic` into a `Seconds` duration fails with the following message:
```
error[E0277]: the trait bound `embedded_time::duration::units::Seconds: core::convert::TryFrom<embedded_time::duration::Generic<<C as embedded_time::clock::Clock>::T>>` is not satisfied
   --> src/mqtt_client.rs:406:83
    |
406 |         let duration: Result<embedded_time::duration::Seconds<u32>, _> = duration.try_into();
    |                                                                                   ^^^^^^^^ the trait `core::convert::TryFrom<embedded_time::duration::Generic<<C as embedded_time::clock::Clock>::T>>` is not implemented for `embedded_time::duration::units::Seconds`
    |
    = note: required because of the requirements on the impl of `core::convert::TryInto<embedded_time::duration::units::Seconds>` for `embedded_time::duration::Generic<<C as embedded_time::clock::Clock>::T>`
```

ryan-summers, 2020-08-10:
Because only the type conversion is not implemented, these can still be used, but much less ergonomically.

```rust
if duration > Generic::new(1, Fraction::new(1, 1)) {
    println!("Greater than 1 second!");
}
```

The above compiles just fine (as expected).

PTaylor-us, 2020-08-10:
```rust
    fn duration_since_with_generic_type<C: time::Clock>(clock: C) {
        let instant1 = clock.try_now().unwrap();
        let instant2 = clock.try_now().unwrap();
        let diff = instant2.checked_duration_since(&instant1).unwrap();

        let secs: Result<Seconds<C::T>, _> = diff.try_into();
        assert!(secs.unwrap() < Seconds(1_u32));
    }
```

```rust
mod clock {
        use embedded_time::Instant;
        use embedded_time::{self as time, fraction::Fraction};

        #[derive(Debug)]
        pub(crate) struct Clock;

        impl time::Clock for Clock {
            type T = u32;
            const SCALING_FACTOR: Fraction = Fraction::new(1, 1_000);

            fn try_now(&self) -> Result<Instant<Self>, time::clock::Error> {
                static mut TICKS: u32 = 0;
                unsafe {
                    TICKS += 1;
                }
                Ok(Instant::new(unsafe { TICKS as Self::T }))
            }
        }
    }
```

Calling the test function with: `duration_since_with_generic_type::<clock::Clock>(clock::Clock);`

ryan-summers, 2020-08-10:
After some discussion with @PTaylor-us, it appears that the issue can no longer be reproduced as originally stated. Concrete durations appear to work with proper `TryInto` conversions. I'm closing this issue as not-an-issue, but will keep watching out for a recurrence so we can potentially figure out what was going on.

closed 2020-08-10 by ryan-summers