FluenTech/embedded-time #79: Breaks docs.rs build

ctron, 2020-10-15:
It looks like the upgrade from Rust nightly 1.48 to 1.49 (which docs.rs uses) breaks the docs build now. To my understanding, this comes from `embedded-time`:

~~~
error[E0284]: type annotations needed: cannot satisfy `<<Clock as Clock>::T as Div<_>>::Output == <Clock as Clock>::T`
   --> /home/jreimann/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-time-0.10.0/./src/instant.rs:182:66
    |
182 |         if add_ticks <= (<Clock::T as num::Bounded>::max_value() / 2.into()) {
    |                                                                  ^ cannot satisfy `<<Clock as Clock>::T as Div<_>>::Output == <Clock as Clock>::T`

error[E0284]: type annotations needed: cannot satisfy `<<Clock as Clock>::T as Div<_>>::Output == <Clock as Clock>::T`
   --> /home/jreimann/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-time-0.10.0/./src/instant.rs:220:66
    |
220 |         if sub_ticks <= (<Clock::T as num::Bounded>::max_value() / 2.into()) {
    |                                                                  ^ cannot satisfy `<<Clock as Clock>::T as Div<_>>::Output == <Clock as Clock>::T`

error[E0284]: type annotations needed: cannot satisfy `<<Clock as Clock>::T as Div<_>>::Output == <Clock as Clock>::T`
   --> /home/jreimann/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-time-0.10.0/./src/instant.rs:273:60
    |
273 |             .cmp(&(<Clock::T as num::Bounded>::max_value() / 2.into()))
    |                                                            ^ cannot satisfy `<<Clock as Clock>::T as Div<_>>::Output == <Clock as Clock>::T`

~~~

PTaylor-us, 2020-10-15:
Thank you for reporting this. #78 was merged to master as a fix, but I haven't released a new version yet.

hannobraun, 2020-11-13:
@PTaylor-us Would you consider releasing a new version some time soon? I have some new libraries that have no working documentation on docs.rs, and other crates that I couldn't release right now if I wanted (as to not break the documentation).

This also causes failure on my CI builds for the nightly channel, and I assume it's going to work its way up to the beta and stable channels soon.

Thank you for your work on this library! I think it's great and fills an important need.

PTaylor-us, 2020-11-13:
@hannobraun Sorry for the delay. I will plan to create a new release in the next day or two.

hannobraun, 2020-11-14:
Thank you, and no need to apologize!

closed 2020-11-15 by PTaylor-us