FluenTech/embedded-time #86: Add implementation for embedded_hal::blocking::delay::Delay trait

Piroro-hs, 2020-11-25:
How about adding `Clock::delay(&self) -> impl Delay` function (maybe behind feature gate)?
I will make a PR if you like.

PTaylor-us, 2020-11-30:
I'm not seeing a `Delay` trait in embedded-hal. It does have `DelayUs` and `DelayMs`. These could possibly be implemented, but I think the proper implementation would be for `Timer`. I'm thinking something like this:

```rust
impl<'a, Clock: crate::Clock, Dur: Duration> embedded_hal::blocking::delay::DelayMs<Clock::T>
    for Timer<'a, OneShot, Armed, Clock, Dur>
{
    fn delay_ms(&mut self, ms: <Clock as crate::Clock>::T) {
        unimplemented!()
    }
}
```

It's been a while since I've worked on this code, but I think we could wrap a timer in the Clock trait.

Piroro-hs, 2020-12-01:
Yes, `DelayMs` and `DelayUs`.

I thought that `Timer` is for constant duration...