FluenTech/embedded-time #88: Add an `integer()` method to `Instant`

korken89, 2020-12-12:
Hi, 

I am trying to convert RTIC to use `embedded-time`, however I need access to the underlying ticks of an `Instant` to be able to set the timer's compare register.
Could this be added?

Thanks!

korken89, 2020-12-13:
I found a work around, but it does not seem like a sane thing to do :)
As the documentation says that `duration_since_epoch` is only "sane" for `Infinite` clocks so using it in this way seems prone to breakage in the future.

```
let integer = instant.duration_since_epoch().integer();
```

PTaylor-us, 2021-03-15:
If I remember correctly, it was an intentional decision to leave this out. The semantics were designed to convert a _generic_ to a _named_ type before doing much of anything with it since the integer count from a _generic_ really has no meaning unless one is converting it manually to a meaningful value which is done automatically when converting to a _named_ type. The _generic_ type was necessary to allow for library users to not be locked into one of the _named_ types for their specific implementation with the intention of converting to a _named_ type as soon as enough context was known.

However, I think I understand the need here and will take a deeper look to, at the very least, refresh my own understanding.

korken89, 2021-03-16:
I'm not quite sure what the `Generic<T>` has to do with it though, as we want to access the ticks of an `Instant`?
The trick here was just that we only have `integer()` implemented for duration (but its a hack that should not be needed IMO): 

```rust
let integer = instant.duration_since_epoch().integer();
```

As long as there is an escape hatch to the underlying `Clock::T`, which can be made to be the hardware timer's tick, I'd say it's fine.
Do you maybe have an alternative approach to this?

PTaylor-us, 2021-03-18:
Apologies. I got mixed up with another issue. It was a similar thought process with Instant in that the value of Clock::T is meaningless unless the underlying implementation of the Clock is known. Can you send me a use-case example to help me wrap my head around it?

korken89, 2021-03-18:
Absolutely!

Here are 2 examples: 
https://github.com/stm32-rs/stm32l4xx-hal/blob/rtic_0.6_monotonic/src/timer.rs#L267
https://github.com/stm32-rs/stm32l4xx-hal/blob/rtic_0.6_monotonic/src/timer.rs#L312

PTaylor-us, 2021-03-19:
If I understand the examples, I don't think I intended for it to be used quite in this way. The Clock interface is for the higher layers as opposed to being used by the lower. In other words, it is intended that the user (as opposed to the implementer) of the Clock trait have no knowledge of the underlying implementation. So I think what I'm saying is that the set_compare shouldn't take an `Instant`. However, even though that was the original intention, that doesn't mean we can't change it. Thoughts?

korken89, 2021-03-20:
This is the very thing that makes this so beautiful.
When the implementer of the clock trait is also the implementer of the actual hardware connection, `embedded-time` gets a true connection to the hardware within that implementation.
This feature is what is being exploited here, and in my opinion its beautiful API design that allows it.

And the final version of this idea you can see here, which shows an API which allows for configuring timers at compile timer:
https://github.com/rtic-rs/dwt-systick-monotonic/blob/master/src/lib.rs

The example before I did not have time to update to use this API, but as you see everything comes together just as one would want.
Hardware understanding at implementation, and not needed from the user side!
So from my view, this should be allowed :) Else we should provide an API to allow it.

PTaylor-us, 2021-04-27:
@korken89 Here's what I'd like to do. Don't add the `integer()` function to Instant, but continue using the `duration_since_epoch()`. I will modify the documentation to define the behavior in a non-infinite context. How does that sound?

korken89, 2021-04-28:
This works just as good to me!
The behavior is well defined, so I think this is the better approach and just adding docs for it will be great.

PTaylor-us, 2021-08-29:
Completed with release of v0.12.0.

closed 2021-08-29 by PTaylor-us