FluenTech/embedded-time #92: Generically take any duration?

korken89, 2021-02-04:
Hi, I have issues with getting the following code to compile as I am unable to get the correct traits needed.
Could you assist here @PTaylor-us ?

```rust
pub fn spawn_after<D: rtic::time::duration::Duration>(duration: D) -> Result<(), ()> {
    let instant = <MyClock as rtic::Monotonic>::now(); // Return an Instant<MyClock>

    // This addition seems to be the issue
    spawn_at(instant + duration, #(,#untupled)*)
}
```

korken89, 2021-02-06:
We found the correct traits that were needed eventually. Would there be any way to make this simpler?

Solution:

```rust
pub fn spawn_after<D>(duration: D) -> Result<(), ()>
    where D: embedded_time::duration::Duration + embedded_time::fixed_point::FixedPoint,
       D::T: Into<<MyClock as embedded_time::Clock>::T>,
{
    let instant = <MyClock as rtic::Monotonic>::now();

    spawn_at(instant + duration, #(,#untupled)*)
}
```

wbuck, 2021-02-13:
I did have a similar request.
Would it be possible to export the `TimeInt` trait?

It would be beneficial when trying to create a custom type, i.e `Rpm`.

PTaylor-us, 2021-04-28:
I want to allow easy generic usage of any `Duration`/`Rate` (including Generic) with only a `embedded_time::Duration` where clause. I'm assuming this is not possible because the `Add` implementation for Instant requires that the Duration impl `FixedPoint`. This will probably take me some time to figure out unless anyone can help me out. I feel like there's a trait pattern that is missing.

PTaylor-us, 2021-08-29:
## Proposed Solution 

- move associated const SCALING_FACTOR out of FixedPoint
- move FixedPoint into Duration 

This should allow this (eliminate the need to bind `FixedPoint`: 
```
pub fn spawn_after<D>(duration: D) -> Result<(), ()>
    where D: embedded_time::duration::Duration,
       D::T: Into<<MyClock as embedded_time::Clock>::T>,
{
    let instant = <MyClock as rtic::Monotonic>::now();

    spawn_at(instant + duration, #(,#untupled)*)
}
```

@korken89 I'm not seeing a good way to eliminate the need for the `Into` binding, though.

PTaylor-us, 2021-08-29:
This may not work fully. I wanted to move the const SCALING_FACTOR out of FixedPoint so Generic could also implement FixedPoint, but obviously, the scaling factor of the FixedPoint number must be accessible by the FixedPoint concrete type.

PTaylor-us, 2021-08-29:
Current plan is to bind FixedPoint to Duration and Rate. Then I will remove the Duration and Rate implementations for the Generic versions.

korken89, 2021-09-02:
Sounds interesting, I'll be lurking. :)

PTaylor-us, 2021-09-12:
Completed for v0.13.0 release PR

closed 2021-09-12 by PTaylor-us