FluenTech/embedded-time #93: Want help maintaining?

korken89, 2021-02-11:
Hey @PTaylor-us !

I understand that you have a lot to do and wonder if you'd like help maintaining this library?
We are moving to `embedded-time` in RTIC and getting better support for more units, and more, is a high priority for us.

Ping me on Matrix or here if you feel like talking!

Best Regards 
Emil

korken89, 2021-03-02:
Ping @PTaylor-us as a reminder to this :)

PTaylor-us, 2021-03-03:
@korken8 I think I need it. Thank you. Ideally, I'd still like to manage the direction of the project and be able to discuss changes before they are merged, but I could definitely use help reviewing the details of PRs and ensuring adequate tests and documentation (including examples if appropriate).

korken89, 2021-03-04:
@PTaylor-us I fully understand! 
I have no aim to change the direction of this library more than to come with proposals and help keep it moving :)
After moving RTIC to `embedded-time` and implementing our [`rtic_monotonic::Monotonic`](https://github.com/rtic-rs/rtic-monotonic/blob/master/src/lib.rs) trait I feel that we have something really good happening here!

I have also experimented with how to set the frequency of HAL timers at compile time (to set the `Fraction`), for example [`dwt-systick-monotonic`](https://github.com/rtic-rs/dwt-systick-monotonic/blob/master/src/lib.rs).
However the `Clock` implementation should probably better live in this library as an example until we have `const generics`.

korken89, 2021-04-26:
Hi @PTaylor-us!

How is this coming a long? 
We are finding more and more issues we need to get fixed in `embedded-time`, and I'm not sure how to proceed.
I can help and review, but I expect in return that PRs and issues move forward.
Preferably, if you feel that you cannot put in the time, could you give me merge and cargo release rights?

If this continues in the current trend I'm starting to feel forced to fork this repository and start something like `embedded-time2` to get the latest updates and bug fixes.
I don't want to sound harsh nor ungrateful, it's just that we are now basing a lot on this library and the functionality it provides.

PTaylor-us, 2021-04-27:
@korken89 I hear you and completely understand if you're feeling frustrated. I have not given these issues the attention they deserve, and you have been exceedingly patient. I'm going to dedicate some time every day to working through these issues. I also want to put together some checklists/templates for PRs to help define what I'm looking for and expecting as well some more high-level documentation concerning what the library is and is not.

korken89, 2021-04-28:
Hi @PTaylor-us,

I really want to thank you for taking the time!
I'll be available for PR generation/reviews and comment ideas as much as I can. Just ping me whenever you feel like it!
Also if there is a feature you feel is needed, but do not have the time - give me a ping there as well and I'll see if I can fix it instead.
This library has removed such a large pain-point that I cannot live without it now :)

closed 2021-08-29 by PTaylor-us