FluenTech/embedded-time #95: Proposal: Only implement `Extension` traits for i32 for ergonomy

korken89, 2021-02-22:
For the sake of ergonomic use I'd recommend that the `Extension` traits only be implemented for `i32` 
so that the expected operation works. Such as `3.seconds()`, `44.milliseconds()` etc.
Getting rid of the must have `_u32` or similar.

What are your thoughts on this @PTaylor-us ?

PTaylor-us, 2021-03-15:
I agree it would be much nicer to use. I struggled with this for a long time trying to make u32 the default, but I couldn't get that to work. I guess, that works for me since 64-bit can still be used, just not with the extensions.

korken89, 2021-03-16:
I made a branch testing this and all worked fine.

I'll prepare a PR after some cleaning!

PR #106 (2021-05-23): Now the extension trait is only implemented for u32
Closes #95
closed 2021-05-25 by korken89