FluenTech/embedded-time #96: Why does the `Clock` Trait require the user to define `const SCALING_FACTOR`?

seanybaggins, 2021-02-24:
Why does the `Clock` Trait require the user to define...
```rust
const SCALING_FACTOR: Fraction
```
? 

Would it be better if `const SCALING_FACTOR` was instead
```rust
fn frequency(&self) -> Rate;
```

**The rational**
Suppose I am writing some `hal` or driver and I would like to implement the `Clock` trait so the user of my `hal`  or driver doesn't need to. Its possible when writing these `hal`'s or drivers that I cannot determine the frequency of hardware I am interfacing with at compile time but can determine the frequency at runtime.

Example of how this may be implemented for a `hal` or driver
```rust
struct StopWatch <F: Rate> {
    inner: Hardware
    frequency: F,
}

impl Clock for StopWatch<F: Rate> {
    ...
    fn frequency(&self) -> Rate {
        self.hardware.get_clock_frequency().MHz()
    }
    ...
}
```

I am still somewhat new to rust in an embedded context so my apologies if this is a horribly misguided suggestion.

korken89, 2021-02-26:
Hi this is to get compile time generation of the conversion math.

To set the speed of a HAL timer at compile time you can see an example here: https://github.com/rtic-rs/dwt-systick-monotonic/blob/master/src/lib.rs

seanybaggins, 2021-04-15:
Thanks for the explanation. Closed the issue.

closed 2021-04-15 by seanybaggins