FluenTech/embedded-time #98: Consider making `TimeInt` public and `FixedPoint::integer` return a value

diondokter, 2021-04-14:
I've done some work on making my existing time constructs compatible with embedded-time.
I ran into two issues however.

Basically, my setup is like this:
```rust
pub struct Time<T: Fixed> {
    secs: T
}
```

The `Fixed` trait comes from the Fixed crate that provides fixed point numbers.

#### Public `TimeInt`:

Since my time struct generic over the type of fixed number I'd like to implement `FixedPoint` for it generically too.
If `TimeInt` were public, then I could implement the `FixedPoint` for any `Fixed` that has a `TimeInt` compatible backing.
So something like:
```rust
impl<T: Fixed> FixedPoint for Time<T>
where
    <T as Fixed>::BITS: TimeInt
{ /* ... */ }
```

#### Return value for `FixedPoint::integer`

The `FixedPoint` trait returns a reference to its presumed underlying value: `fn integer(&self) -> &Self::T;`
But because my underlying is not an int, this can't work. To get the integer value I need to call the `to_bits` method.
```rust
fn integer(&self) -> &Self::T {
    &self.secs.to_bits()
}
```
This creates a temporary that can't be returned as a reference.

Perhaps I don't understand the intricacies, but I don't think there will be (much of a) performance impact to just do some copying here.

Anyways, these two change would improve the ergonomics for me considerably. It's a big breaking change though...

hannobraun, 2021-04-28:
I've also run into a case where I need to specify `TimeInt` in a where clause, but can't.

PTaylor-us, 2021-04-28:
@korken89 What do you think about this? I have no problem with making TimeInt public, but the `FixedPoint::integer` change to return a value rather than a reference may have some nuance that I'm not seeing being a little "rusty" on Rust. (I'm sure no one has used that one before ☺)

hannobraun, 2021-04-29:
`FixedPoint::T` is `TimeInt`, which in turn is always `Copy`. I don't think this proposed change will have any adverse side effects (other than being a breaking change).

PTaylor-us, 2021-04-30:
I'm looking into this some more. It looks like every `integer()`-like function I have is returning a reference due to this sematic being common among other crates I used. I'd like to keep things consistent to reduce _surprises_, but also want this to work for you. At the moment I'm considering changing all `integer()`-like functions to a value return.

korken89, 2021-05-04:
Generally, when returning something that is `Copy` and does not need to keep the lifetime of from where it came, it makes most sense to not have it be a reference.

diondokter, 2021-05-11:
Both issues have been solved by #101 and #103

closed 2021-05-11 by diondokter