FluenTech/embedded-time #99: Should the Clock trait take a &mut self instead of &self?

seanybaggins, 2021-04-15:
It would be nice if `try_now` would update the internal state of the clock before attempting to return the time. As it currently stands, it is up to the user of clock to make sure it is updated before they attempt to call `try_now`. Thoughts?

PTaylor-us, 2021-08-29:
Sorry for the ridiculously long delay in responding. I'm really hoping to get back into a pattern of working on this crate.

In my implementations, I have been reading directly from hardware which doesn't necessitate changing the internal state of the Clock object. Do you have an implementation that would require that?

eldruin, 2021-08-29:
Here is some relevant discussion about this: https://github.com/FluenTech/embedded-time/issues/32

PTaylor-us, 2021-08-29:
@eldruin Thanks for the reminder. It's been awhile.

eldruin, 2021-08-29:
You are welcome Peter! nice to see you again :)