FluenTech/embedded-time #102: remove generic Timer struct's State parameter

PTaylor-us, 2021-04-28:
Ref: #75 

The State type parameter means that a Timer that is going to change state cannot be (nicely) stored in a struct.

Does the Type type parameter also block some use cases?

korken89, 2021-05-04:
I'll give this a look.

korken89, 2021-05-27:
Hi, I have given this a look and we stand between 2 choices:

Either one has type states which is nice for compile time checking but cumbersome to use in runtime and place in structs,
or we go with the state being an enum in the timer itself which means runtime checking of states and we loose the compile time checks.

One can also wrap the timer object in an enum, where each enum variant maps to a state.
However this would be a bit weird as one turns the state problem inside out with making state explicit to the user to handle.
Eg `TimerState::Running(timer)`, where the user has to check and do the right thing depending on the `TimerState`.

What are your thoughts on the @PTaylor-us ?

PTaylor-us, 2021-05-30:
@korken89 I think the issues brought up with the current type-state scheme are serious enough to switch over to runtime checks. I like the elegance of the type states, but it's just not a good fit for this situation. So I think storing an enum state in the timer is a good way to go.

PTaylor-us, 2021-08-15:
@korken89 Have you already done any work on this? I may want to work on it, but don't want to duplicate effort.

korken89, 2021-08-15:
Unfortunately not, I became a dad recently so my contributions to open source projects is currently on hold for a while longer. :)