FluenTech/embedded-time #104: smarter macros for defining intra-unit conversions

PTaylor-us, 2021-05-06:
There are many instances when converting from a larger-magnitude unit with width u32 to a smaller-magnitude unit with width u64 could use `From`/`Into` rather than their fallible forms. However, my current level of macro expertise limits me to either one or the other (fallible or infallible).

korken89, 2021-05-27:
One option is to move the macro generation to an internal proc-macro.
This kind of change would be trivial then.

PTaylor-us, 2021-05-30:
I'm not very familiar with proc macros. Is this something you are capable of doing, @korken89?

korken89, 2021-05-30:
I have no problems with proc-macros as with RTIC I mostly write them :) But I am not familiar with all the generation that the current macros are performing.
I'll study the current macro to get a better feel for it.