FluenTech/embedded-time #125: Remaining/Elapsed timer calculation for generic clock

marius-meissner, 2022-02-24:
The following example is failing by unsatisfied trait bounds:
````Rust
use embedded_time::{Clock};
use embedded_time::duration::{Microseconds};

fn test<C: Clock>(clock: C)
{
    let duration = Microseconds::new(100);
    let timer = clock.new_timer(duration).start().unwrap();
    timer.remaining();
}
````

I already tried replicating some trait bound from remaining() to test() method, without success.

I am pretty sure that this is not a fault of the library. Perhaps the documentation could therefore be extended to include a few generic examples.

Anyway, thanks for this awesome library!

Version: v0.12.1 @ rustc 1.60.0-nightly (88fb06a1f 2022-02-05)

rmja, 2023-01-06:
Year, I have the same problem...

rmja, 2023-01-06:
I guess you need:

```rust 
fn test<C: Clock>(clock: C) where C::T: Into<u32> {
}